SmartGridready/SGrJava #51: Errors when trying to build SGrJava from source

mkrebs81, 2024-04-16:
I have tried to build the _SGrJava_ project from source on Windows, but I ran into problems during the generation of the _emfEI4Modbus_ classes and the Gradle build.

## My Setup

I have followed the instructions in the README:

First, I installed Eclipse for Java developers 2024-03 into `C:\Users\matt\eclipse\java-2024-03\eclipse` and added the EMF / XSD frameworks. I also installed the Adoptium JDK 21.0.2.

The Git repositories are checked out under `D:\dev`, to their `master` branch:

- `D:\dev\SGrSpecifications`
- `D:\dev\SGrJava`
- `D:\dev\SGrJavaDrivers`

The Gradle configuration in `C:\Users\matt\.gradle\gradle.properties` contains:

```
xsd2java.eclipse.home=c:/users/matt/eclipse/java-2024-03/eclipse
xsd2java.xsd.resource=d:/dev/SGrSpecifications/SchemaDatabase/SGr/SGrIncluder.xsd
```

## The issues

I have successfully built and published (to mavenLocal) the _GenDriverAPI_ and _EasyModbus_ from `SGrJavaDrivers`.
However, when I navigate to `SGrJava` and run `gradlew.bat clean build`, the build fails:

The process creates a directory `D:\emfEI4Modbus`, which seems to be wrong, because it is outside of the project root. If I understand correctly, it should be at least under `D:\dev`.

Also, the `emfEI4Modbus\src` directory is not copied to `SGrJava\InterfaceFactory\emfEI4Modbus\src`, this is why the build fails due to missing Java classes.
When I copy `src` manually and run `gradlew.bat build`, the build succeeds.

How can this be fixed?

ergo-furrer, 2024-04-16:
Hi Matthias, I found the problem:

Before code-generation there must already exist an eclipse workspace (.metadata) directory in your projects root folder, in your case D:\dev\.metadata. The workspace must already contain a project named 'emfEI4Modbus', otherwise a new workspace is created and put into the root folder d:\emfEI4Modbus. 

As workaround you can fix this as follows:

1. Delete your current .metadata folder.
2. Run Eclipse and select the workspace folder `d:\dev` --> now you should have a new workspace folder '.metadata' in `d:\dev`
3. In Eclipse select [File][Import...]
4. In the Import Wizard select [Gradle][Existing Gradle Project] and then [Next]
5. Under Project Root Directory [Browse] to `d:\dev\SGrJava` --> now you have an existing emfEI4Modbus project at the desired location.
6. Now you can do `d:\dev\SGrJava> gradlew clean build` again

Unfortunately we were not aware of this problem because we all already had existing Eclipse workspaces when doing the first gradle build.

I will think of a solution for this problem and hope you get all running for now using the proposed workaround. Please tell me if you get stuck at some place.

Regarding Java 21: Indeed we need at least Gradle 8.4. We also think of upgrading the Gradle Version.

ergo-furrer, 2024-04-18:
@mkrebs81 I was trying to compile with a Java Version above 1.8 and found a problem with the Eclipse EMF code generator.
Invalid code is generated for Java versions above 1.8. I fixed the script that generates the code and merged it to master.
I checked the build with Java 17. 
See [Pull request 54](https://github.com/SmartGridready/SGrJava/pull/54)

mkrebs81, 2024-04-18:
Thank you. I tried it yesterday using your instructions, and it worked! At least I was able to build the library from within Eclipse and publish the package to my local Maven repo.

I just tried it again, after pulling the newest changes from master.
I deleted the whole workspace, created it again, deleted all generated files and imported the SGrJava Gradle project again.

Unfortunately everything is messed up now.
The projects are no longer imported as multiple projects, I only see the SGrJava project.
The project also shows Java-SE 1.8 as JRE and complains about unsupported class versions.
Cleaning the project does not help, and I can't run any Gradle tasks.

ergo-furrer, 2024-04-18:
Hi Matthias, sorry for that.

We decided today to go at least to Java 11!

ergo-furrer, 2024-04-18:
This was a result of a discussion with Arnd and we suppose that embedded systems (like Raspberry Pi) support at least Java 11 for now.

closed 2024-05-27 by mkrebs81