SmartGridready/SGrJava #65: Add support for string value mappings on REST-API datapoints.

ergo-furrer, 2024-06-07:


mkrebs81, 2024-06-12:
PoC implementation in Branch https://github.com/SmartGridready/SGrJava/tree/rest-api-post-formdata

ergo-furrer, 2024-06-12:
Thanks for the Info, I will have a look.

mkrebs81, 2024-06-14:
I just have tested my implementation with the Shelly Gen1 local API.
I can set a relay to true/false using GET requests.

- data point type = Relais, but with boolean type
- mapping defined --> true = on, false = off

so far that seems to work.

the POST requests using form data do not work correctly yet, I will conduct more testing today.

mkrebs81, 2024-06-14:
I was able to fix the POST request content type handling in my latest commit.

I will now look into the "URIBuilder". that looks promising regarding query and form parameters.

mkrebs81, 2024-06-17:
I have created a PoC implementation in https://github.com/SmartGridready/SGrJava/tree/rest-api-query-params

It uses my suggested specification changes in https://github.com/SmartGridready/SGrSpecifications/tree/rest-query-params
and should solve URL encoding, query and form parameters at once.

@ergo-furrer @ArndGerns do you like this approach?

ergo-furrer, 2024-06-17:
Hi Matthias, I think it's the right approach to define a query and form parameters as a key-value list in the XML-Schema.

I had a brief look at SGrJava.ApacheRestServiceClient. I think the method `getFormParameters()` and its dependents are obsolete now and we can remove it.

mkrebs81, 2024-06-17:
Thanks for the feedback.
I will add a commit that removes the old behaviour.

closed 2024-06-18 by mkrebs81