SmartGridready/SGrJava #76: Refactoring and removal of legacy functionality

mkrebs81, 2024-06-25:
Add topics about the next big refactoring here.

## Separation of Java Maven packages

I suggest a slightly different separation of Maven packages:

- `sgr-specification` contains the classes generated from XML schema and the description loader
- `sgr-driver-api` contains Java interfaces for all kinds of drivers (Modbus, REST, MQTT, Contact)
    - driver or client interfaces
    - factory interfaces
- `sgr-drivers` contains default driver implementations (EasyModbus, also REST, MQTT etc.)
- `sgr-library` contains the SGr java library with device interfaces and implementations

## Namespace

- use a consistent namespace for Java code, e.g. `ch.smartgridready.***`

please add your ideas here...

mkrebs81, 2024-06-28:
I am going to have a meeting with David and Robin today, where we are going to have a look at the current state of the **Python** library. I have found a lot of differences to the Java library in terms of architecture and usage.

Perhaps it would be a good idea to discuss and design the general architecture and interfaces together, for _both_ Java and Python.
The result after refactoring would be a more consistent developer experience.
I will talk about this on the next roundtable meeting as well.

ergo-furrer, 2024-06-28:
@mkrebs81 will you meet Robin in person or do you have teams meeting, so i could attend.

mkrebs81, 2024-06-28:
it's only in person today. I will keep you informed on our findings.

ergo-furrer, 2024-06-28:
@mkrebs81 ok, thanks!

ChrisBroenni, 2024-06-28:
Good idea. We may have to have an on site meeting to follow up on this architecture level issues.
As one of the architects I like to be part of this discussion,

Please look also at the following issues:
ABB RTU meter did not work on Java test anymore (did not yet dived in)
An ABB  Modbus TCP does NOT exist, I expect that there is a layer disruption porblem for using an EID for proxy configuration (?).
An EID is always meant for one TransPortlayer, TranportLayer span has always to be managed by a router or communicator.
The proxy should follow this aechitecture. Otherwise, we can not scale up systems. But its only on  a rumor basis done thinking.

I'll look at  this until tuesday, I'll also check for renamed attributes as the initially followed the rachecturaö logi of follwing exiting standards (like EEBUS / IEC)trhe target.  The SGCP group comes up with an issue to look at EEBUS again.

mkrebs81, 2024-07-16:
@ergo-furrer @ArndGerns 
I suggest you complete the "simple" refactoring first, before changing any functionality:
- namespaces, directory structure, naming of classes
- package build for specification
- move all driver interfaces and factory interfaces to driver api package

I will not do any development on the library until then, because that would interfere with your refactoring.
I will gladly give feedback if desired.

closed 2024-08-09 by ergo-furrer