SmartGridready/SGrJava #82: Version detection with palantir fails

mkrebs81, 2024-07-02:
When I check out the master branch and compile, I get `1.1.0-SNAPSHOT` as version string, which seems correct.
This also happens when the commit of the master branch matches the latest release tag.

When I check out a release tag like `1.1.0` directly, I get `unspecified` as version string, instead of `1.1.0`.

ergo-furrer, 2024-07-02:
Hmm, that's strange. Let me check this. Palantir should take the latest tag, which is 1.1.0, when you check out 1.1.0 I think.
However may I misunderstood something...

mkrebs81, 2024-07-02:
When I add `version = rootProject.version` to the CommHandler4Modbus build file, it works.

Setting the version in `publishing {...}` is probably not correct, or does not work as intended.

ergo-furrer, 2024-07-02:
To check, I did the following:
- $ git branch -d master (delete local master)
- $ git fetch --all --tags
- $ git checkout tags/1.1.0 -b master  (must be named 'master' otherwise it is a SNAPSHOT)'

then I did:
- $ ./gradlew printVersion  --> output is 1.1.0

It seems to work on my machine without moving the tag out of the `publishing {...}`. 

However, I am not sure if it was a good idea to add SNAPSHOT depending on the branch not being 'master'. Maybe using just 'isCleanTag' and 'commitDistance' is enough. So we can use the branch name 'rel-1.1.0' or something like that whe checking out the 1.1.0 tag.

ergo-furrer, 2024-07-02:
Right. I will remove the `master` check. I will remove the check.

mkrebs81, 2024-08-19:
will be removed in refactoring

closed 2024-08-19 by mkrebs81