SmartGridready/SGrJava #83: Add support for build pipelines using github actions.

ergo-furrer, 2024-07-03:


mkrebs81, 2024-07-16:
A suggestion regarding the CI pipeline and Eclipse:

You can check out multiple repositories in a common workspace directory, see https://github.com/actions/checkout#Checkout-multiple-repos-side-by-side

Run the Eclipse container with the workspace directory mounted in the container.

Call the appropriate shell script from within the container. The entry point should check for the mounted directory and then call the script.
Or use CMD instead of ENTRYPOINT, if you want to specify the command to execute on the commandline.

ergo-furrer, 2024-07-18:
The CI pipeline for SGrJava works now. However I did add the changes on branch #76 and not #86 .

mkrebs81, 2024-07-18:
@ergo-furrer thanks. the branch #76 is totally fine! #86 should be tackled at a later point in time, when everything is refactored and the Nexus repository works.

I think the pipeline is a good first step.
We probably need two pipelines in the future:

1. gets triggered from specification repo and only builds the sgr-specification package.
2. get triggered locally and only builds the sgr-java package, referencing the external sgr-specification package.

ergo-furrer, 2024-07-22:
@mkrebs81 Yes, it's the plan plan to have 2 pipelines. I will change this after the refactoring ist done. However I wanted to have the proof that the code generation with eclipse, using Github actions, does work. So I am relieved now 😅 .

mkrebs81, 2024-08-19:
I'm trying to split the pipeline in two now.

- no root build script, only specific build scripts in CommHandler and SGrSpecification
- Specification build uses SGrSpecifications repository and SGrJava
- CommHandler build only uses SGrJava and references sgr-specification on Maven repository

closed 2024-10-01 by mkrebs81