SmartGridready/SGrJava #89: Use XJC to generate Java classes of SGr specification

mkrebs81, 2024-08-19:
Using Eclipse/EMF to generate classes is quite tedious.

Using XJC can be much easier, because it has a Gradle plugin (https://github.com/bjornvester/xjc-gradle-plugin) and does not need an installation of the Eclipse IDE.

The downside is that all code which accesses the EMF classes needs to be adapted.

mkrebs81, 2024-08-20:
### Advantages

- class generation is **MUCH** easier than using Eclipse EMF

### Disadvantages

- no singleton _V0Factory_, _ObjectFactory_ instead
- no _V0Package_ with data type constants
- XML loader needs modification

mkrebs81, 2024-08-20:
it seems that XJC does not like enum types that have values starting with a digit...

found this workaround: https://stackoverflow.com/questions/30370456/generating-big-enum-classes-with-numbers-in-enum-values-jaxb-xjc

it works, but the generated names are `VALUE_1`, `VALUE_2` etc.

mkrebs81, 2024-08-20:
`RestApiDataPointConfiguration` is not generated correctly --> properties reference to JAXB elements instead of the correct classes

mkrebs81, 2024-08-20:
the whole `EStructualFeature` stuff does not exist in JAXB. getting type names dynamically may get complicated.

ergo-furrer, 2024-09-18:
The code using JAXB works now.
However I found that there is a dependency to in the sgr-driver-api that uses generated classes (MQTT-interface). I think I will remove that. 
To come over conflicts with the EMF i did temporarily publish JAXB versions of the libraries e.g.  1.1.0-JAXB-SNAPSHOT ....

mkrebs81, 2024-09-23:
Thank you. I did not have time to look at it last week, but I added some modifications today:
1. parametrization of device descriptions, so test do not fail
2. added JAXB-SNAPSHOT to SGrJava as well, so I can distinguish versions
3. added an issue and branch to SGrJavaDrivers, so I can compile with JAXB and publish to local repo

I have noticed one issue with the JAXB implementation:
The current "0.3.0" version of the ABB B23 EID cannot be loaded - unfortunately I have not yet found the issue; the exception is unclear...

mkrebs81, 2024-09-23:
I was able to fix the EID. somehow `<generalRemarks>...</generalRemarks>` was the problem, even though other EIDs also have this element.

mkrebs81, 2024-09-30:
IMHO it is ready to be integrated now.

closed 2024-09-30 by mkrebs81