SmartGridready/SGrJava #107: Discuss implementation of data points with multiple values

mkrebs81, 2024-10-09:
## Discussion

According to the specification, data points can have multiple values (array) by specifying `<arrayLength>{number}</arrayLength>`, where all array values are of the data type specified in the data point (see below).
There are not many EIDs which require arrays, but this may change in the future.

Currently there is no _generic_ way of accessing data point array values:
- the Python library does not support it at all
- the Java library supports a separate `getValArr()` and `setValArr()` method, but only for Modbus devices

**As discussed in the roundtable meeting on 2024-10-08, this issue is the basis for further discussion.**

## Current State

### Data Types

Data points support the following data types:
- boolean
- integer (8/16/32/64 bit, signed/unsigned)
- float (32/64 bit)
- string
- datetime
- enum (literal, encoded as string)
- json (object, encoded as string)
- bitmap (object, encoded as string)

Some of these types have not been fully implemented:
- datetime (difficult to convert from device values, as there is no standard format)
- json (should probably be implemented as json node or dynamic object)
- bitmap (should probably be implemented as actual map/dict)

## Possible solution approaches

### Treat everything as array

The easiest solution: everything is an array, in most cases of size 1. `getVal()` and `setVal()` take an array of values.

However, this may be tedious for developers, as most data points will be of size 1.

### Introduce array type

This will probably make things even more complicated, because the array still needs a regular data type for its values.
Would need to overhaul the type system and `Value` classes completely.

**please feel free to contribute to this issue**

ergo-furrer, 2024-10-22:
I have a proposal regarding the handling of array values.

- The existing GenDeviceApi does not need getValArray()  anymore.
- Existing code must not be changed (except the code that uses getValArray())
- There is a new ArrayValue class the wraps the array of single Values
- Maybe we should also an isArray() method to the Value class...

Below is an example on how to use it:

```java

@ExtendWith(value = MockitoExtension.class)
public class ArrayValueTest {

    @Mock
    private GenDeviceApi deviceApi;

    @Test
    void arrayValue() throws Exception {

        when(deviceApi.getVal(any(), any())).thenReturn(createArrayValue());

        // API usage:
        // Return the value as regular value type
        Value value = deviceApi.getVal("VoltageAC", "VoltageAC_L1");

        // Even if it is an ArrayValue underneath, we still can read the first value of the array as usual:
        assertEquals(1.24f, value.getFloat32());

        // Usage of value as array:
        Value[] arrayValue = value.asArray(); // Will return only the first element for primitive types
                                              // Will return all values for array types

        // Values as float
        assertEquals(1.24f, arrayValue[0].getFloat32());
        assertEquals(2.34f, arrayValue[1].getFloat32());

        // Values as double[]
        double[] doubles = Arrays.stream(arrayValue).mapToDouble(Value::getFloat64).toArray();
        assertEquals(1.24f, doubles[0]);
        assertEquals(2.34f, doubles[1]);

        // Values as string
        assertEquals("1.24", arrayValue[0].getString());
        assertEquals("2.34", arrayValue[1].getString());
    }

    private Value createArrayValue() {

        var val0 = Float32Value.of(1.24f);
        var val1 = Float32Value.of(2.34f);

        // new value type, extends from value
        return ArrayValue.of(val0, val1);
    }
```

ergo-furrer, 2024-10-22:
@mkrebs81 I could implement that if you agree. At first only modbus will support arrays but the we have the same API for all interfaces....

mkrebs81, 2024-10-22:
> @mkrebs81 I could implement that if you agree. At first only modbus will support arrays but the we have the same API for all interfaces....

Sounds good. Can you implement a proof of concept after we finish migration of the driver api code?

ergo-furrer, 2024-10-22:
> Sounds good. Can you implement a proof of concept after we finish migration of the driver api code?

Yes I can do so.

closed 2024-10-24 by ergo-furrer