SmartGridready/SGrJava #116: Use ServiceLoader for driver implementations

mkrebs81, 2024-10-21:
I suggest using the _ServiceLoader_ (see https://www.baeldung.com/java-spi for example) to use the concrete driver implementations.

The approach works like SLF4j for example, where you only define a specific implementation of the logging facility, and it is automatically found in the classpath.

This would completely decouple the driver implementation and would allow communicator developers to use their own implementation (if desired).

ergo-furrer, 2024-10-22:
@mkrebs81 
Yes we should move the code first.  I am convinced to move the GenDriverAPI to SGrJava.
I am a bit reluctant with the EasyModbus driver. It's not 'SGrJava core code' from my point of view. Other 3rd parties could provide additional code for their drivers within this project. I might be good to have them separated from the core library.

mkrebs81, 2024-10-22:
> @mkrebs81 Yes we should move the code first. I am convinced to move the GenDriverAPI to SGrJava. I am a bit reluctant with the EasyModbus driver. It's not 'SGrJava core code' from my point of view. [...]

Yes, you are right. I would leave the `EasyModbus` project in the `SGrJavaDrivers` repository.
And in the future add the HiveMQ and Apache clients to the `SGrJavaDrivers` repository.

mkrebs81, 2024-10-23:
@ergo-furrer I have implemented a PoC in the attached branch.
It works quite well!
However, since it requires additional interfaces and some modifications, I would suggest that we publish some bugfixes first, and integrate this in a release 2.1 or so.

ergo-furrer, 2024-10-24:
@mkrebs81 OK, fine for me.

ergo-furrer, 2024-10-24:
@mkrebs81 

There might be a problem within the `SGrModbusGatewayFactory`.

```java
line 39:
      case TCP:
                try {
                    transport = createRTUTransport(interfaceDescription.getModbusRtu());
                } catch (NoSuchElementException e) {
                    throw new GenDriverException("No Modbus factory implementation found on classpath");
                }
                transport = createTCPTransport(interfaceDescription.getModbusTcp());
                break;
```
Can't we try createTCPTransport directly?  / First assignment to transport seems to be useless (overwritten anyway)

mkrebs81, 2024-10-24:
I will look into that transport problem.
Perhaps we can make the whole factory simpler, because the transport factory interface was added to the driver API.

I did not improve that, because it was still a PoC

mkrebs81, 2024-10-24:
BTW, when I am going to fully implement it, I will also move the HTTP and messaging implementation to the driver repository.

ergo-furrer, 2024-10-24:
@mkrebs81 Everything fine, just was interested in how it's done now...

mkrebs81, 2024-10-26:
@ergo-furrer I have implemented part of it, but I still need to improve the Modbus driver handling and the separation of the MQTT driver. I won't be in office next week, but I might do some work from time to time...

closed 2024-11-28 by mkrebs81