SmartGridready/SGrJava #120: Check Data Type Conversion between Modbus and Data Point

mkrebs81, 2024-10-23:
Data points of Modbus devices have 2 data types:

1. the type of the data inside Modbus registers
2. the generic data type of the exposed data point

It seems that reading Modbus data returns a subclass of `Value` according to Modbus, but is not always converted to the external data type when it does not match the Modbus data type.

mkrebs81, 2024-10-23:
@ArndGerns @ergo-furrer my modification seems to fix the issue when the DP type is boolean but Modbus type is not.
I'm not sure if the fix is in the correct place, because other DP types might need conversion, too.

ergo-furrer, 2024-10-23:
@mkrebs81 for conversion of numerical values from modbus to boolean, it is thought to use the 'ModbusBoolean' type. If the value received from modbus is to be interpreted as boolean we can configure the type of the modbus datapoint as follows:

```xml

 <modbusDataType>
                  <boolean>
                    <trueValue>255</trueValue>
                  </boolean>
  </modbusDataType>
```
```xml

 <modbusDataType>
                  <boolean>
                    <falseValue>255</falseValue>
                  </boolean>
  </modbusDataType>
```

you can use any numeric value to either define true or false.

This is already implemented, so to fix the problem I suggest to fix the EI-XML. 

Sorry that I did not remember this feature, when we discussed the problem yesterday....

mkrebs81, 2024-10-23:
> @mkrebs81 for conversion of numerical values from modbus to boolean, it is thought to use the 'ModbusBoolean' type. If the value received from modbus is to be interpreted as boolean we can configure the type of the modbus datapoint as follows:
> 
> ```
>  <modbusDataType>
>                   <boolean>
>                     <trueValue>255</trueValue>
>                   </boolean>
>   </modbusDataType>
> ```
> 
> ```
>  <modbusDataType>
>                   <boolean>
>                     <falseValue>255</falseValue>
>                   </boolean>
>   </modbusDataType>
> ```
> 
> you can use any numeric value to either define true or false.
> 
> This is already implemented, so to fix the problem I suggest to fix the EI-XML.
> 
> Sorry that I did not remember this feature, when we discussed the problem yesterday....

I agree on the first part - this was also my first solution.
My assumption was that the data types of the DP itself and the Modbus configuration should match.

- Is there any case where the DP and Modbus do _not_ match?
- I also assume that, when a DP has a certain data type, I should always receive an appropriate subclass instance of Value. Is that correct?

ergo-furrer, 2024-10-23:
> I agree on the first part - this was also my first solution. My assumption was that the data types of the DP itself and the Modbus configuration should match.

At least the data-types must be convertable from modbus to the generic format. The boolean is somehow a special case as it needs extra information to be converted  in some cases.

> 
> * Is there any case where the DP and Modbus do _not_ match?

They might be different, for example generic INT64 / modbus INT32. 

> * I also assume that, when a DP has a certain data type, I should always receive an appropriate subclass instance of Value. Is that correct?

The Value subclass of the modbus datatype is returned. Conversion is done when reading the value in the specific data format, Int32Value.getInt64() for example. If the value cannot be converted (e.g. out of range) an Exception is thrown.

mkrebs81, 2024-10-24:
I think we can leave it for now. As long as the data types match, there won't be a problem.

I had some issues reading the different data types, because I use `getString()` in the testlab software.
Its result of depends on the actual subclass.
I probably can work around it by using `getBoolean` etc. and then convert to string.

closed 2024-10-24 by mkrebs81