SmartGridready/SGrJava #133: Unable to parse FP XML

mkrebs81, 2024-12-09:
As mentioned in the last meeting I have been experimenting with a FP XML loader in order to validate EIDs against the declared FPs.

However, I always get a parse exception, probably due to some schema validation issue.

Any ideas what could be the cause?

ChrisBroenni, 2024-12-09:
In heat pump EID's  the reason was the missing brackets in  {{tcp_address}}  fixed in  1_IBT_Jaxb branch

( pending pull request in SGrSecification)

mkrebs81, 2024-12-09:
This issue is not about loading EIDs, but about loading FP descriptions (from `FuncProfiles` directory).

AFAIK the FPs do not have placeholders which could cause schema validation issues.

ergo-furrer, 2024-12-09:
@mkrebs81 I was able to fix this issue. FP-XML can be loaded now.

mkrebs81, 2024-12-09:
> @mkrebs81 I was able to fix this issue. FP-XML can be loaded now.

awesome! I'm going to test that as soon as I can.

closed 2025-01-06 by mkrebs81