SmartGridready/SGrJava #134: Catch NULL values in messaging protocol

mkrebs81, 2024-12-09:
## Situation

A messaging (MQTT) device receives data on a subscribed data point.

When the message transformation results in a string value of `"null"`, e.g. when the requested JSON key has no value or actually has null as value, the data point value cannot be updated due to conversion error.

The error message is `For input string: "null"`, indicating that it could not be converted to a float, int or whatever.

## Options

a) keep exception
b) check for NULL value and do not execute callback/consumer in case of a NULL value

mkrebs81, 2024-12-09:
@ergo-furrer the current state seems to work more or less in the testlab environment.
When you have some time left, please have a look at it.

ergo-furrer, 2024-12-16:
@mkrebs81 
Indeed, the Problem arises whenever the received value cannot be converted to the data Format expected by the Generic interface (e.g. Numbers or Enums). 

Probably it is a good idea to add some response-validation to the SGrMessagingDevice implementation in order to report the cause of the error more accurately such as "invalid message received, received value is not a number...".

Regarding, whether we let the user know when invalid data is received or we just skip the received message: 
I think the user should get to know when invalid data is received.

- When reading a value synchronously getVal(): it would be suitable to throw an exception, however i think it is better when getVal() throws and not Value.getFloat32() for example (means some additional validation within the SGrMessagingDevice).

- When reading asynchrounously using subscribe() we should callback with an Either.left() that contains the Exception containing the validation error. It is up to the client-user then to decide how to handle the Error (or ignore it ...)

- Regarding caching the last read value:  I think we should not chache the invalid value (currently the invalid value is cached).