SmartGridready/SGrJava #144: ModbusExceptionCode handling may block runtime code execution

ChrisBroenni, 2024-12-20:
ModbusExceptionCodes may be blocked not retirnung Get and Set calls.
All ModbusExceptionCode should generate wornings when returning  from calling setters and getters

**ModbusExceptionCode**
   enumeration value="IllegalFunction""
    **enumeration value="IllegalAddress"**
    **enumeration value="IllegalDataValue"**
    enumeration value="SlaveFailure"
    enumeration value="ACK"
    enumeration value="SlaveBusy"
    enumeration value="NACK"
    enumeration value="MemoryParityErr"
    enumeration value="GtwyPathErr"
    enumeration value="GtwyTargetErr"

Issue observed using SGr_04_0020_xxxx_Heliotherm_HeatPumpV0.2.1.xml ,     HeatPumpBase:HPOpModeCmd

mkrebs81, 2025-01-06:
I looked at this issue, and the current state is that this specification feature is not implemented at all in the commhandler library.

I also do not fully understand the documentation of the `AccessProtectionEnabled` in the XSD...

I am not even sure if Modbus exception codes can actually be propagated from the driver to the commhandler (I believe you can catch subclasses of `de.re.easymodbus.exceptions.ModbusException`, but these do not convey an exception code)

ergo-furrer, 2025-01-07:
I had a look at the EasyModbus implementation. Only the error codes 01-04 are handled. All other codes are ignored.
All Exceptions listed below derive from `de.re.easymodbus.exceptions.ModbusException` and are wrapped as cause of a `GenDriverModbusException`  with a more or less suitable error message and a log entry. However there is no Exception code within the Exception.  --> Exception mapping: see `ModbusCallHandler` in EasyModbus project.

Code | Name | Description| EasyModbus throws
-- | -- | -- | --
01 | Illegal Function | The function code in the request is not supported by the server. | FunctionCodeNotSupportedException
02 | Illegal Data Address | The data address specified in the request is invalid or unavailable on the server.| StartingAddressInvalidException
03 | Illegal Data Value | The value in the request is invalid or outside the allowable range for the addressed data.|QuantityInvalidException
04 | Slave Device Failure | An unrecoverable error occurred while the server was processing the request.|ModbusException (read error)
05 | Acknowledge | The server has accepted the request and is processing it, but it will take some time to complete.| not handled
06 | Slave Device Busy | The server is engaged in processing a long-duration task and cannot handle the request right now.|not handled
08 | Memory Parity Error | The server detected a parity error while accessing internal memory.|not handled
0A | Gateway Path Unavailable | The gateway is unable to establish a connection to the target device.|not handled
0B | Gateway Target Device Failed to Respond | The target device did not respond to the gateway within the expected time.|not handled

We could add handling for the additional exception codes in the Rossmann driver code. And also add the Modbus exception code to the `GenDriverModbusException`.

What do you think?

mkrebs81, 2025-01-07:
We could add those exception codes; it does not look like a lot of effort...

But I would still discuss the specification first - to understand what it means.

ChrisBroenni, 2025-01-13:
The main problem was that it was blocking execution of the code  (may be that was I puzzled by the  for me new debugger view?)

mkrebs81, 2025-02-10:
The new j2mod based driver is the new default since CommHandler release 2.2.0.

We may need to check if the new driver has better exception handling.
Propagating the exception codes to the commhandler would still need modifications.

closed 2025-10-23 by mkrebs81