SmartGridready/SGrJava #145: Value conversion problem

ChrisBroenni, 2024-12-20:
2 datapoints with exactly the same setting return a  differnet value with exactly same content (proven with second client)

 INFO  c.s.c.m.i.HeatPumpTester - HeatPump HelioTCP
 INFO  c.s.c.m.i.HeatPumpTester -   HeatPumpBase:    HPOpModeCmd=HEL_HP_OFF/0,  HPOpStateAsEnum=HEL_HP_OFF/0,  ErrorFlagSGr=false,  OutsideAirTemp=0.0°C
 INFO  c.s.c.m.i.HeatPumpTester -     ** SupplyWaterTemp=-50.0°C,  ReturnSupplyWaterTemp=23.2 °C** 

Issue observed using SGr_04_0020_xxxx_Heliotherm_HeatPumpV0.2.1.xml , HeatPumpBase:SupplyWaterTemp, InputRegister @12 
                                                                                                                        HeatPumpBase:ReturnSupplyWaterTemp, , InputRegister @13

mkrebs81, 2025-02-04:
Note to myself: test this when we receive the Heliotherm WP controller

mkrebs81, 2025-03-17:
Not a Modbus issue - was caused by errors in EID!

closed 2025-03-17 by mkrebs81