SmartGridready/SGrJava #146: enumeration does not send correct value in Modbus

ChrisBroenni, 2024-12-20:
an enumeration data point does not send correct value, always reeds 0

   dataPointName HPOpModeCmd
        literal                                         /      ordinal
        HEL_HP_OFF                              / 0
        HEL_HP_AUTOMATIC                / 1
         HEL_HP_SUMMER_MODE        /  3
         HEL_HP_COOLING_OP             /  2
         HEL_HP_DURATION_OP           / 4
         HEL_HP_REDUCED_OP             / 5
         HEL_HP_VACATION_MODE      / 6
         HEL_HP_PARTY_MODE             / 7
         HEL_HP_OUTHEATING_MODE / 8
        HEL_HP_DSO_BLOCK                / 9
         HEL_HP_MAIN_SWITCH           /10 

Issue observed using SGr_04_0020_xxxx_Heliotherm_HeatPumpV0.2.1.xml , HeatPumpBase:HPOpModeCmd

procedure: reproducing of the behaviour in embedded environment
does missalignemnt of the number sequence create this problem?

mkrebs81, 2025-01-06:
I tested the EID with the current commhandler 2.1.2 and a software-defined Modbus TCP device , where I can read/write arbitrary values into the registers.

Prerequisites:
- DP enum values must have both literal _and_ ordinal defined, or the mapping may not work

The results:
- after I write a literal and read again, I get the correct literal/ordinal from the device
- after I write an ordinal and read again, I get the correct literal/ordinal from the device
- the sequence of literals/ordinals does not matter, provided that there are no conflicts

mkrebs81, 2025-02-04:
Note to myself: check again in test lab when we receive the Heliotherm WP controller.

mkrebs81, 2025-03-17:
Not a Modbus issue - was caused by errors in EID!

closed 2025-03-17 by mkrebs81