SmartGridready/SGrJava #149: Handle JSON data type in both directions

mkrebs81, 2025-01-27:
## Situation

The Java library does not support the `json` data type yet.
It will soon be required in SGCP and dynamic tariff applications.

The first step should be the implementation of JSON values as strings, which could later be changed to actual JSON node objects.

Writing JSON data also needs an implementation which supports transformations using JMES mappings etc.

## Tasks

- [ ] implement `JsonValue` class for _reading_ data points and return as string
- [ ] implement

mkrebs81, 2025-01-27:
After talking to @kevin-arm I have found out that this topic is still in discussion because of the requirements of dyn. tariff providers, SGCP etc.

Because things like "sub-data-points" are not supported by the specification, and we probably cannot change that on short notice, I suggest the following approach, which we can implement without further changes to the specification:

- when a DP has the `json` data type, return the value as `JsonNode`, so library users can decode the object as they wish
- creation of JSON node happens _after_ JMES path filter or mapping
- when working with JSON data internally (e.g. in REST), work with JSON nodes as long as possible, and only convert to string when necessary

mkrebs81, 2025-02-11:
@ergo-furrer @ArndGerns I have been able to improve the PoC implementation.
The JSON conversion is a bit more efficient, and I also added some test cases. I think we can do some real testing soon.

ergo-furrer, 2025-03-12:
@mkrebs81 @ArndGerns 

We could also add some functionality to bind a Value to a given Object (i assume that there are Python libs the allows similiar object bindings):

Let's say we have a tariffs class like
```java

   private static class Tariff {
        public String startTimestamp = "2024-01-01T01:45:00+0100";
        public String endTimestamp =   "2024-01-01T01:45:00+0100";
        public TariffRecord[] tariffs =
                {   new TariffRecord("normal",new BigDecimal("25.1"), "Rp./kWh"),
                    new TariffRecord("eco", new BigDecimal("28.2"), "Rp./kWh"),
                };
    }
```

If we like, this `Tarif` class could even be defined in the functional profile XSD and therefore be part of the specification. The class can then be generated by JAXB.

The usage will be simple:

```java

   @Test
    void createJsonValueFromStringAndReadIntoObject() throws Exception {

        ObjectMapper objectMapper = new ObjectMapper();
        String json = objectMapper.writeValueAsString(new Tariff());

        Value givenValue = JsonValue.of(json);

        Tariff returnVal = givenValue.getJson(Tariff.class);

        assertEquals(new BigDecimal("25.1"), returnVal.tariffs[0].tariff);
        assertEquals(new BigDecimal("28.2"), returnVal.tariffs[1].tariff);
    }

    @Test
    void createJsonValueFromObjectAndReadiIntoObject()  {

        Value value = JsonValue.of(new Tariff());

        Tariff returnVal = value.getJson(Tariff.class);

        assertEquals(new BigDecimal("25.1"), returnVal.tariffs[0].tariff);
        assertEquals(new BigDecimal("28.2"), returnVal.tariffs[1].tariff);
    }

```

The JsonValue implementation just uses the Jackson ObjectMapper and will be straight forward:
(Code added to the current JsonValue in the branch related to this issue, `value` is still a JsonNode)

```java

   // Read
   @Override
    public <T> T getJson(Class<T> aClass) {
        ObjectMapper mapper = new ObjectMapper();
        try {
            return mapper.treeToValue(value, aClass);
        } catch (JsonProcessingException e) {
            var msg = String.format("Unable to map JSON string '%s' to the given class '%s'", value, aClass.getSimpleName());
            throw new IllegalArgumentException(msg);
        }
    }

   // Create
  public static JsonValue of(Object object) {
        ObjectMapper objectMapper = new ObjectMapper();
        return new JsonValue(objectMapper.valueToTree(object));
    }

   //
    public static JsonValue of(String string, Class aClass) throws Exception {
        ObjectMapper objectMapper = new ObjectMapper();
        return new JsonValue(objectMapper.readTree(string));
    }

```

mkrebs81, 2025-03-12:
Interesting idea.
I think we could add your code to my implementation as an optional convenience method, while keeping the generic JsonNode in the JsonValue class.

The library users will still be able to decode the JsonValue into whatever they want.

Returning the Tariff class directly would have some downsides IMHO:
- works only in this specific use case
- how would this work in other programming languages?
- how do we know which class to use, without additional info in the EID?

mkrebs81, 2025-03-12:
I removed the task "JSON schema in EID" because we had decided only to document the schema in the EID, but not fully specify it in the data type definition.

mkrebs81, 2025-03-13:
@ergo-furrer I added your suggested method extension to my implementation. looks good to me.

closed 2025-03-18 by ergo-furrer