SmartGridready/SGrJava #154: Evaluate ReadTheDocs as documentation tool

ergo-furrer, 2025-02-10:
ReadTheDocs has been proposed as documentation tool. 

- ReadTheDocs supports Sphinx which is more powerful then markdown and allows more structure in the documents.
- It supports generating documentation from in-code documentation like PythonDoc and JavaDoc
- ReadTheDocs has its focus on doc generation from Python which is straight forward. For Java we have to find a suitable solution to integrate with ReadTheDocs.

mkrebs81, 2025-02-17:
Maybe a tool which generates a static website from text files, Python docs or Java docs could be a solution.
The resulting site could be hosted via Github pages and would not require an other hosting platform, registration etc.

Examples:

- [Jekyll](https://jekyllrb.com/) with [javadoc support](https://github.com/fennifith/mdjavadoc)

mkrebs81, 2025-05-20:
We use Gradle to build the javadoc code documentation, and publish that on Github pages.

closed 2025-05-20 by mkrebs81